bit-bots/bitbots_main #19: Unified formatting, docs, variables, signatures

JanGut, 2019-08-01:
For formatting we should use the PEP8 standard.

Besides that we should decide for one of the variations of type documentation for methods as well as whether we should use _underscores for "private" variables or not.

timonegk, 2019-08-01:
I think that conventions for code documentation (e.g. function parameters, types, etc.) are also important to ease the automatic generation of a code documentation via sphinx or doxygen.

ftsell, 2019-08-01:
>  to ease the automatic generation of a code documentation via sphinx or doxygen.

exactly. For example if we were to use doxygen style docstrings in python we could use doxygen to document our python code as well. The benefit is that doxygen makes it far easier to link into existing documentation and therefore crossreference to other packages

ftsell, 2019-08-14:
I've just looked into ReadTheDocs and here's my summary:

- ReadTheDocs is in essence just a hosting platform for automatic sphinx building. It creates it's documentation with Sphinx
- The look and feel of ReadTheDocs is just a Sphinx template
- I found a way to generate C++ documentation with doxygen and then convert the discovered information into sphinx (though I don't know how well that works in our case)

It should therefore be possible to use Sphinx and Doxygen for code generation and have all information properly extracted. We can than use Sphinx to display our doc in a unified way (even fancy like ReadTheDocs). Then we can host that generated documentation somewhere of our choosing

SammyRamone, 2019-08-15:
some of the documentation for software is currently in bitbots_meta/doc (e.g. for QuinticWalk). This should maybe changed, so that the documentation is always in the corresponding package and only referenced

SammyRamone, 2019-08-15:
when changing to a new doc system the "how to doku manual" also will have to be updated. it is currently outdated anyway, but correcting this now would probably be senseless

SammyRamone, 2019-08-15:
also make sure that including images will work with new doc system, since it is currently broken (see e.g. quintic walk documentation)

timonegk, 2019-09-25:
We settled on using the Google Code Style for Python and the ROS code style for cpp. Docstring in python are formatted using ReST syntax. Further information will be provided in the documentation.

closed 2019-09-25 by timonegk