bit-bots/bitbots_main #78: make pull-all continue after error

SammyRamone, 2020-10-12:
Currently the "make pull-all" script exits if one repository could not be pulled. it would be nice to continue with the rest and just print a summary which repos could not be pulled at the end.

jaagut, 2022-06-22:
@val-ba and I tried solving this issue, but it's off course not as simple as it seems.
Our main idea was running additional commands inside the `git submodule foreach`-loop with and `&&`

- Writing `$sm_path` in a temporary var (relative path of current submodule)
- Running `git pull`
- Clear the temporary variable

If the pull failed, the temporary variable would not have been cleared, therefore we know the submodule, where pull failed. We can then append the temporary variable to a list/string, that collects all failed submodule to summarize in the end.

Problems with this approach:

- Keeping `set -e` in the script: If pull fails, everything quits, and we cannot run additional commands, such as printing a summary
- Without `set -e`: Pull requires human input (editor opens to commit a merge), we don't want interruptions
- Some pull options might suppress opening the editor (e.g. `dry-run`, but not tested), but in that case, how do we know the pull did not run successfully?

Do you have any ideas on how to solve this?

Flova, 2022-06-22:
I am also a bit critical towards the continue in error thing as people might not notice an error and end up with a mismatched meta. This is especially the case for new members.

Flova, 2022-06-22:
And my git and shell magic is not strong enough to posses the powers to help you in this case.

jaagut, 2022-06-23:
@timonegk What is your opinion? Is it worth proceding or do we close this?

timonegk, 2022-06-23:
I think this would be entirely possible, but I am not sure if this is really necessary or helpful. Especially so close to the competition our focus should be on fixing and testing instead of implementing trivial convenience features, in my opinion.

closed 2023-08-31 by jaagut