bit-bots/bitbots_main #137: Fix quintic walk build

texhnolyze, 2022-07-08:
`bitbots_quintic_walk` build issues:
```
--- stderr: bitbots_moveit_bindings
CMake Error at CMakeLists.txt:15 (find_package):
  By not providing "Findbio_ik_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "bio_ik_msgs", but CMake did not find one.
```
Due to #f363f15 in bitbots_meta, where `bio_ik_service` was removed, which contained the `bio_ik_msgs` module.

SammyRamone, 2022-07-08:
True we do need the bio_ik_msgs package. So it should go back into meta. But we should also remove the other packages in this git since they are not building https://github.com/bit-bots/bio_ik_service/pull/2

jaagut, 2022-07-09:
Resolved by #139

closed 2022-07-09 by jaagut