bit-bots/bitbots_main #177: Predeclare and validate parameters by YAML

texhnolyze, 2023-03-08:
As in ros2 parameters have to be predeclared we want to configure them by YAML configuration while allowing validation and bounds, by utilizing: https://github.com/PickNikRobotics/generate_parameter_library

While the library generates C++ classes for python we need to implement our own approach, by potentially generating a class/object at runtime.

The follwoing steps must be done:
- introduces the `generate_parameter_library` lib
- implement code/object generation for python
- setup YAML configuration of parameters for all modules in `bitbots_meta`
- adjust code of modules to predeclare parameters utilizing this approach
- rm backwards compatible flags `allow_undelcared_parameters` and `declare_parameters_from_overrides` from all modules

jaagut, 2023-10-02:
From our meeting notes: Use only one YAML file if only one parameter set exists

Originally posted: https://github.com/bit-bots/bitbots_meta/issues/204#issue-1880466698

jaagut, 2023-10-02:
This tool has already been introduced in the `bitbots_template_repository`. However the following parameter-files need to be replaced:

84 files: 
- [ ] `bitbots_behavior/bitbots_body_behavior/config/animations.yaml`
- [ ] `bitbots_behavior/bitbots_body_behavior/config/body_behavior.yaml`
- [ ] `bitbots_lowlevel/bitbots_buttons/config/buttons.yaml`
- [ ] `bitbots_lowlevel/bitbots_ros_control/config/analyzers.yaml`
- [ ] `bitbots_lowlevel/bitbots_ros_control/config/pressure_amy.yaml`
- [ ] `bitbots_lowlevel/bitbots_ros_control/config/pressure_converter.yaml`
- [ ] `bitbots_lowlevel/bitbots_ros_control/config/pressure_melody.yaml`
- [ ] `bitbots_lowlevel/bitbots_ros_control/config/pressure_nobot.yaml`
- [ ] `bitbots_lowlevel/bitbots_ros_control/config/pressure_rory.yaml`
- [ ] `bitbots_lowlevel/bitbots_ros_control/config/wolfgang.yaml`
- [ ] `bitbots_misc/bitbots_basler_camera/config/camera_settings.yaml`
- [ ] `bitbots_misc/bitbots_basler_camera/config/default.yaml`
- [ ] `bitbots_misc/bitbots_ceiling_cam/config/camera_settings_ceiling_cam.yaml`
- [ ] `bitbots_misc/bitbots_extrinsic_calibration/config/amy.yaml`
- [ ] `bitbots_misc/bitbots_extrinsic_calibration/config/default.yaml`
- [ ] `bitbots_misc/bitbots_extrinsic_calibration/config/donna.yaml`
- [ ] `bitbots_misc/bitbots_extrinsic_calibration/config/jack.yaml`
- [ ] `bitbots_misc/bitbots_extrinsic_calibration/config/melody.yaml`
- [ ] `bitbots_misc/bitbots_extrinsic_calibration/config/rory.yaml`
- [ ] `bitbots_misc/bitbots_ipm/config/soccer_ipm.yaml`
- [ ] `bitbots_misc/bitbots_teleop/config/controller.yaml`
- [ ] `bitbots_misc/bitbots_utils/config/game_settings.yaml`
- [ ] `bitbots_misc/bitbots_utils/config/global_parameters.yaml`
- [ ] `bitbots_misc/system_monitor/config/config.yaml`
- [ ] `bitbots_motion/bitbots_dynamic_kick/config/kick_config.yaml`
- [ ] `bitbots_motion/bitbots_dynamic_kick/config/kick_sim_config.yaml`
- [ ] `bitbots_motion/bitbots_dynup/config/dynup_optimization.yaml`
- [ ] `bitbots_motion/bitbots_dynup/config/dynup_robot.yaml`
- [ ] `bitbots_motion/bitbots_dynup/config/dynup_sim_darwin.yaml`
- [ ] `bitbots_motion/bitbots_dynup/config/dynup_sim.yaml`
- [ ] `bitbots_motion/bitbots_hcm/config/hcm_wolfgang.yaml`
- [ ] `bitbots_motion/bitbots_odometry/config/odometry_config_amy.yaml`
- [ ] `bitbots_motion/bitbots_odometry/config/odometry_config_donna.yaml`
- [ ] `bitbots_motion/bitbots_odometry/config/odometry_config_jack.yaml`
- [ ] `bitbots_motion/bitbots_odometry/config/odometry_config_melody.yaml`
- [ ] `bitbots_motion/bitbots_odometry/config/odometry_config_rory.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/deep_quintic_pybullet_wolfgang.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/deep_quintic_robot.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/deep_quintic_webots_chape.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/deep_quintic_webots_mrl_hsl.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/deep_quintic_webots_op3.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/deep_quintic_webots_rfc.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/deep_quintic_webots_wolfgang.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/optimization.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_bez_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_chape_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_gankenkun_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_mrl_hsl_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_nao_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_nugus_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_op3_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_rfc_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_robotis_op2_optimized_single_steps.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_robotis_op2_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_sahrv74_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_wolfgang_evaluation.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_wolfgang_robot_no_limits.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_wolfgang_robot.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_wolfgang_simulator.yaml`
- [ ] `bitbots_motion/bitbots_quintic_walk/config/walking_wolfgang_viz.yaml`
- [ ] `bitbots_motion/bitbots_rl_motion/config/rl_walk_sim.yaml`
- [ ] `bitbots_navigation/bitbots_localization/config/config.yaml`
- [ ] `bitbots_navigation/bitbots_localization/config/fields/bangkok/config.yaml`
- [ ] `bitbots_navigation/bitbots_localization/config/fields/feldraum/config.yaml`
- [ ] `bitbots_navigation/bitbots_localization/config/fields/gazebo/config.yaml`
- [ ] `bitbots_navigation/bitbots_localization/config/fields/spl/config.yaml`
- [ ] `bitbots_navigation/bitbots_localization/config/fields/webots/config.yaml`
[... 16 lines cut ...]
- [ ] `wolfgang_robot/wolfgang_robocup_api/scripts/start.sh`

The following files need to be adapted too:
- [ ] `bitbots_lowlevel/bitbots_ros_control/src/pressure_converter.cpp`
  212:     e<< YAML::Key << "ros__parameters";

- [ ] `bitbots_misc/bitbots_utils/bitbots_utils/game_settings.py`
  112:     ros_parameters = config["parameter_blackboard"]["ros__parameters"]
[... cut ...]
- [ ] `humanoid_league_misc/humanoid_league_game_controller/README.md`
  5:   ros__parameters:

- [ ] `lib/imu_tools/imu_filter_madgwick/launch/imu_filter_component.launch.py`
  18:     # This code will load the parameters from the yaml (removing the namespace/nodename/ros__parameters heading) so
  21:         params = yaml.safe_load(f)['imu_filter']['ros__parameters']

- [ ] `wolfgang_robot/wolfgang_robocup_api/scripts/start.sh`
  127:   ros__parameters:
  
Originally posted: https://github.com/bit-bots/bitbots_meta/issues/204#issuecomment-1719456764

Flova, 2023-10-07:
> This tool has already been introduced in the `bitbots_template_repository`. However the following parameter-files need to be replaced:

Not all parameter files need to be replaced tho! Only one template is needed for a given parameter set. If two (e.g.) sim and real or different robots (like here https://github.com/bit-bots/bitbots_motion/pull/423) are present a single template can be used to declare parameters whose values are loaded the normal way from the old style parameter file. Normally both files would be needed ether way (one for the template and one for the actual parameter set), but this can be simplified into one file if we utilize the default value option in the declaration process. This is also possible if params are declared by code (the way without the lib).  But by including them in the template only a single file is needed. Normal parameters files are still needed if multiple sets are present.