bit-bots/bitbots_main #214: Installation of latest basler drivers is not handled on robots

texhnolyze, 2023-11-16:
# Summary
With the new `make_basler.sh` script we now have to manually copy and execute the script to
install the latest basler deb packages.

## Expected behavior
To prevent confusion and version drift I would expect the latest balser package to be installed after a robot setup.

## Current behavior
The script is not executed on the robot and if we have updated the used pylon driver version in the meantime, it will not be updated unless done manually.

## Possible Solution
It would be a lot easier to include the execution of the script in a remote shell, with our `deploy_robots.py` script.
Another alternative could also be the installation of the pylon drivers via `ansible`, during the setup of ros.

jaagut, 2023-11-24:
I would prefer the installation via ansible.

jaagut, 2024-01-26:
This now gets handled by the deploy tool

jaagut, 2024-02-11:
Resolved with #223

closed 2024-02-11 by jaagut