bit-bots/bitbots_main #267: Vision dies

Flova, 2024-01-23:
# Summary

```
[taskset-3] Traceback (most recent call last):
[taskset-3]   File "/home/bitbots/colcon_ws/install/bitbots_vision/lib/bitbots_vision/vision", line 33, in <module>
[taskset-3]     sys.exit(load_entry_point('bitbots-vision', 'console_scripts', 'vision')())
[taskset-3]   File "/home/bitbots/colcon_ws/build/bitbots_vision/bitbots_vision/vision.py", line 184, in main
[taskset-3]     rclpy.spin(node)
[taskset-3]   File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/__init__.py", line 228, in spin
[taskset-3]     executor.spin_once()
[taskset-3]   File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 725, in spin_once
[taskset-3]     raise handler.exception()
[taskset-3]   File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/task.py", line 239, in __call__
[taskset-3]     self._handler.send(None)
[taskset-3]   File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 424, in handler
[taskset-3]     await call_coroutine(entity, *arg)
[taskset-3]   File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 348, in _execute_subscription
[taskset-3]     await await_or_execute(sub.callback, *args)
[taskset-3]   File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/executors.py", line 108, in await_or_execute
[taskset-3]     return callback(*args)
[taskset-3]   File "/home/bitbots/colcon_ws/build/bitbots_vision/bitbots_vision/vision.py", line 141, in _image_callback
[taskset-3]     self._run_vision_pipeline(image_msg)
[taskset-3]   File "/home/bitbots/colcon_ws/build/bitbots_vision/bitbots_vision/vision.py", line 166, in _run_vision_pipeline
[taskset-3]     self._run_components(image_msg)
[taskset-3]   File "/home/bitbots/colcon_ws/build/bitbots_vision/bitbots_vision/vision.py", line 177, in _run_components
[taskset-3]     vision_component.run(image_msg)
[taskset-3]   File "/home/bitbots/colcon_ws/build/bitbots_vision/bitbots_vision/vision_modules/yoeo/vision_components.py", line 87, in run
[taskset-3]     self._yoeo_instance.predict()
[taskset-3]   File "/home/bitbots/colcon_ws/build/bitbots_vision/bitbots_vision/vision_modules/yoeo/yoeo_handlers.py", line 154, in predict
[taskset-3]     self._create_detection_candidate_lists_from(detections)
[taskset-3]   File "/home/bitbots/colcon_ws/build/bitbots_vision/bitbots_vision/vision_modules/yoeo/yoeo_handlers.py", line 165, in _create_detection_candidate_lists_from
[taskset-3]     self._det_candidates[self._det_class_names[int(detection[5])]].append(c)
[taskset-3] IndexError: list index out of range
[ERROR] [taskset-3]: process has died [pid 14825, exit code 1, cmd 'taskset -c 6,7 /home/bitbots/colcon_ws/install/bitbots_vision/lib/bitbots_vision/vision
```

## Expected behavior
 No exception

## Current behavior
It dies

## Steps to Reproduce
Run the [2023_07_07_flo_yoeo_light/](https://data.bit-bots.de/models/2023_07_07_flo_yoeo_light/) yoeo on the robot for some time.

phinik, 2024-02-22:
> The https://data.bit-bots.de/models/2023_07_07_flo_yoeo_light/model_config.yaml is missing the goal post class.

What do you mean? Is this supposed to be the reason?

Flova, 2024-03-18:
It is. We need to adjust the model config. The vision code is fine.

closed 2025-06-03 by Flova