bit-bots/bitbots_main #311: Maybe it is better to change the initial value

Bozenton, 2024-01-23:
Hi there, 
In the file of `motion_odometry.cpp`, I noticed that you give `-1` to `current_support_state_` and `previous_support_state_` as initial value: 
https://github.com/bit-bots/bitbots_motion/blob/0a908c8b19d1252ee87680fe2f45cfe24a922ef6/bitbots_odometry/src/motion_odometry.cpp#L23-L24

but both of them are defined as `char` variables:
https://github.com/bit-bots/bitbots_motion/blob/0a908c8b19d1252ee87680fe2f45cfe24a922ef6/bitbots_odometry/include/bitbots_odometry/motion_odometry.h#L20-L21

As far as I know, different compilers may address `char` in different ways, some deal with it as signed, some deal with it as unsigned: 
https://stackoverflow.com/questions/2054939/is-char-signed-or-unsigned-by-default

For example, in my jetson nx board, it thinks the `char` is `unsigned char`. So when it is initialized with `-1`, then it is stored as `255`, which would cause bug for this line: 
https://github.com/bit-bots/bitbots_motion/blob/0a908c8b19d1252ee87680fe2f45cfe24a922ef6/bitbots_odometry/src/motion_odometry.cpp#L174
While compiler in my PC thinks it as `signed char` and thus nothing wrong happens. 

So I think maybe it is better to use a different initial value for these two variables, or change the type of them?

Flova, 2024-06-19:
We could also declare them as std::optional and make them explicitly nullable that way.

Flova, 2024-06-27:
Closed as part of said pr.

closed 2024-06-27 by Flova