bit-bots/bitbots_main #320: Walking: tune walk kicks

jaagut, 2024-01-23:
## Is your feature request related to a problem? Please describe.
We cannot kick at the moment.

## Describe the solution you'd like
We should use the walk-kick feature instead.
Don't do this by moving the foot further and back again after touching the ball, but rather setting the foot down after accelerated step-movement.
Side-kicks are also possible by rotating the foot. Afterward, we may need to revert the foot rotation.
Look at Hulks or other SPL teams for details.

## Describe alternatives you've considered

00lake00, 2024-02-18:
### Meeting with Flo and Jan

* lower frequency of step and phase reset would be helpful in keeping balance @Flova 
* kick_phase parameter should be 0.4/0.5
* velocity nice for longer kick
* roll for longer kick?
* longer foot down length for kick than normal walking? --> balance issue for big step after kick but with small difference nice

*TODO*
* parameter tuning on real roboter
* plotyuggler show joint position error

timonegk, 2024-05-02:
Walk kick should be adapted to the current kick animation

PR #537 (2024-07-14): Fix walk kick and utilize it
# Summary

Changes the walk kick splines so it works on the real robot (mostly). I also integrated it into the behavior so it is utilized for kicking. This is not perfect yet, but I will further tune it on the real field tomorrow. I also tuned the localization and fixed a few minor things there to get an overall better performance.

## Related issues
Cleaned up version of #525 
Closes #320 

## Checklist

- [x] Run `colcon build`
- [ ] Write documentation
- [ ] Test on your machine
- [x] Test on the robot
- [x] Create issues for future work
- [x] Triage this PR and [... cut]
closed 2024-07-14 by Flova