bit-bots/bitbots_main #364: ros control driver dies during startup

Flova, 2024-03-20:
# Summary
```
taskset-10] free(): invalid pointer
[taskset-10] Stack trace (most recent call last) in thread 1736426:
[taskset-10] #18   Object "", at 0xffffffffffffffff, in
[taskset-10] #17   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7ff04552684f]
[taskset-10] #16   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7ff045494ac2]
[taskset-10] #15   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7ff0458dc252, in
[taskset-10] #14   Object "/opt/ros/iron/lib/librmw_cyclonedds_cpp.so", at 0x7ff0441c01c2, in
[taskset-10] #13   Object "/opt/ros/iron/lib/librmw_cyclonedds_cpp.so", at 0x7ff0441c5408, in
[taskset-10] #12   Object "/opt/ros/iron/lib/x86_64-linux-gnu/libddsc.so.0.10.4", at 0x7ff044113faa, in dds_take
[taskset-10] #11   Object "/opt/ros/iron/lib/x86_64-linux-gnu/libddsc.so.0.10.4", at 0x7ff044113a0b, in
[taskset-10] #10   Object "/opt/ros/iron/lib/x86_64-linux-gnu/libddsc.so.0.10.4", at 0x7ff044104354, in
[taskset-10] #9    Object "/opt/ros/iron/lib/x86_64-linux-gnu/libddsc.so.0.10.4", at 0x7ff044104286, in
[taskset-10] #8    Object "/opt/ros/iron/lib/x86_64-linux-gnu/libddsc.so.0.10.4", at 0x7ff044100de1, in
[taskset-10] #7    Object "/opt/ros/iron/lib/x86_64-linux-gnu/libddsc.so.0.10.4", at 0x7ff044100645, in
[taskset-10] #6    Source "./malloc/malloc.c", line 3391, in __libc_free [0x7ff0454a5452]
[taskset-10] #5    Source "./malloc/malloc.c", line 4439, in _int_free [0x7ff0454a2a43]
[taskset-10] #4    Source "./malloc/malloc.c", line 5664, in malloc_printerr [0x7ff0454a0cfb]
[taskset-10] #3    Source "../sysdeps/posix/libc_fatal.c", line 155, in __stack_chk_fail [0x7ff045489675]
[taskset-10] #2    Source "./stdlib/abort.c", line 79, in abort [0x7ff0454287f2]
[taskset-10] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7ff045442475]
[taskset-10] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[taskset-10]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[taskset-10]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7ff0454969fc]
[taskset-10] Aborted (Signal sent by tkill() 1736335 1000)
[ERROR] [taskset-10]: process has died [pid 1736335, exit code -6, cmd 'taskset -c 0 /home/bitbots/colcon_ws/install/bitbots_ros_control/lib/bitbots_ros_control/ros_control --ros-args --params-file
```

texhnolyze, 2024-07-19:
After fixing a couple of valgrind issues and the codestyle in `bitbots_ros_control`, by not manually using `new`, `malloc` and `free` this has not been happening anymore.
Can we close this issue?

timonegk, 2024-07-20:
I would close it because it did not happen again in the last months.

closed 2024-07-20 by timonegk