bit-bots/bitbots_main #366: Limit diagonal speed in path planning

Flova, 2024-03-23:
# Summary

Currently we can go full speed forwards and sideways at the same time. I think it would be reasonable to limit that based on the maximum forward/sideward movement. I don't know if a kite or ellipse shape for the maximum speed envelope would be suited best.

## Is your feature request related to a problem?

Sometimes steps seem unreasonably large when we go diagonally. I observed this on the real robot at the integration test.

## Describe the solution you'd like

Not only limit the movement speed along the x and y direction (while keeping the direction). We can also interpolate the max speed based on the angle.

## Describe alternatives you've considered

Lower the x and y speed. We accept slower speeds.

texhnolyze, 2024-06-11:
@Flova Isn't this already done by these [lines](https://github.com/bit-bots/bitbots_main/blob/main/bitbots_navigation/bitbots_path_planning/bitbots_path_planning/controller.py#L126-L129)?
Can this issue be closed?

Flova, 2024-06-11:
I think the current max speed envelope looks a bit like this:
[image]

The Idea was something like this:
[image]

because we might not be able to walk with the same max vel diagonally than forward. But we need to chack that.

PR #534 (2024-07-14): Limit diagonal speed in path planning
# Summary
Closes #366 

## Checklist

- [x] Run `colcon build`
- [x] Write documentation
- [x] Test on your machine
- [x] Test on the robot
- [x] Create issues for future work
- [x] Triage this PR and label it
closed 2024-07-14 by Flova