bit-bots/bitbots_main #442: Ignore usb ethernet interface in system monitor

Flova, 2024-05-06:
# Summary
Currently, our diagnostics state is on ERROR most of the time. This is due to the USB network interface getting stale after being disconnected at the start of the game. This results in a stale system metric, and the diagnostics aggregator sets the overall system state on ERROR.

## Describe the solution you'd like
- Add the ability to ignore a specific interface in the system monitor

## Describe alternatives you've considered
- Don't monitor system interfaces
- Keep it as it is and just ignore it

jaagut, 2024-07-19:
Why did you reopen this issue?

Flova, 2024-07-25:
I think she reopened it because we just did the quick fix by ignoring all network monitoring instead of blacklisting a specific device.