bit-bots/bitbots_main #530: Replace `mimic3` with [`piper`](https://github.com/rhasspy/piper)

texhnolyze, 2024-07-14:
# Summary
As stated in https://github.com/bit-bots/ansible/pull/14 and https://github.com/bit-bots/bitbots_main/pull/529, the `mimic3` company mycroft is defunct and does not supply updates to the tts system anymore, and we even had to adjust some code to use different URLs for downloads.

## Describe the solution you'd like
Replace setup and usage of `mimic3` with [`piper`](https://github.com/rhasspy/piper) in `bitbots_tts`, `anisble` and `requirements/robot.txt`.

## Describe alternatives you've considered
Alternatively, we could look at other available tts systems.

val-ba, 2025-04-10:
Maybe https://github.com/coqui-ai/TTS is an alternative?

val-ba, 2025-04-10:
Is the speak_ip.py used and if so where?