bit-bots/bitbots_main #635: ros control should only publish joint states after the wait time is elapsed

Flova, 2024-12-12:
# Summary
Currently, joint states are published as soon as we get data from the motors. But we write data to the motors after a fixed delay to make sure they are fully booted. This delay should also apply for the publishing of data as otherwise there is a period of time where we get data from the motors (signaling to other parts of the software that they are available) but we can not control them yet. This is also related to the startup delay in the HCM which could be removed in this case. See #631 for the current approach.

val-ba, 2025-04-17:
Would you do that by just adding a bus_start_time_ condition as in wolfgang_hardware_interface.cpp in line 308, but just for the read method aswell? Isnt there other information which is also read over the bus? Should we wait for that aswell?

Flova, 2025-04-19:
Ideally we should wait with everything. I am not 100% sure how I would implement this feature. I think it should be a few more lines tho .

jaagut, 2025-11-22:
@Flova @val-ba Why is this "in progress"? Is there any change/PR associated to this? Is this still relevant for a non-Wolfgang future?