bit-bots/bitbots_main #655: deploy_robots.py cannot remove build folder

val-ba, 2025-01-20:
# Summary

When trying to use the deploy_robots.py script with the following command `./scripts/deploy_robots.py donna -sb -p bitbots_quintic_walk` the build directory cannot be removed.
The command run is rm -rf ~/colcon_ws/build ~/colcon_ws/install ~/colcon_ws/log' exited=1> and if you run it directly on the robot it does not cause an error.
## Expected behavior

It should just remove the directory.
## Current behavior

This is the error message: 

rm: cannot remove '/home/bitbots/colcon_ws/build/biped_interfaces'
: Directory not empty
Traceback (most recent call last):
  File "/homes/17vahl/colcon_ws/src/bitbots_meta/scripts/deploy/tasks/build.py", line 55, in _clean
    results = connections.run(cmd_clean, hide=hide_output())
  File "/homes/17vahl/.local/lib/python3.10/site-packages/fabric/group.py", line 116, in run
    return self._do("run", *args, **kwargs)
  File "/homes/17vahl/.local/lib/python3.10/site-packages/fabric/group.py", line 282, in _do
    raise GroupException(results)
fabric.exceptions.GroupException: {<Connection host=172.20.1.14 user=bitbots>: <UnexpectedExit: cmd='rm -rf ~/colcon_ws/build ~/colcon_ws/install ~/colcon_ws/log' exited=1>}

## Steps to Reproduce

1. Run ./scripts/deploy_robots.py donna -sb -p bitbots_quintic_walk

## Possible Solution

Flova, 2025-01-20:
Note this only happens sometimes

val-ba, 2025-04-17:
I also had this issue on my machine, it might not be due to the script, but general ubuntu things.

Flova, 2025-06-03:
This seems like an upstream issue.