bit-bots/bitbots_main #660: Localization, path_planning, behavior fail on non `--symlink-install` build

texhnolyze, 2025-02-27:
# Summary
Our codebase is not executable when running `colcon build` without the parameter `--symlink-install`.

**This was an issue on our Ubuntu 24.04/Jazzy branch, but I assume it is also an issue in general**

## Expected behavior
Be able to execute our whole codebase with the bringup launch files:

- `rl bitbots_bringup simulator_teamplayer.launch`
- `rl bitbots_bringup teamplayer.launch`

without any modules crashing.

## Current behavior
When runnning the code base with bringup a couple of modules crash, due to import errors and `Nodes` not being available.

## Steps to Reproduce

1. Build the code base only with `colcon build`
2. Execute a bringup script (sim or not)
3. Localization, path_planning, behavior fail to executed due to import issues and follow up issues of `Nodes` not being available

### Error log output
```
[path_planning-27] Traceback (most recent call last):
[path_planning-27]   File "/homes/21griepen/colcon_ws/install/bitbots_path_planning/lib/bitbots_path_planning/path_planning", line 33, in <module>
[path_planning-27]     sys.exit(load_entry_point('bitbots-path-planning==0.0.0', 'console_scripts', 'path_planning')())
[path_planning-27]              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[path_planning-27]   File "/homes/21griepen/colcon_ws/install/bitbots_path_planning/lib/bitbots_path_planning/path_planning", line 25, in importlib_load_entry_point
[path_planning-27]     return next(matches).load()
[path_planning-27]            ^^^^^^^^^^^^^^^^^^^^
[path_planning-27]   File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
[path_planning-27]     module = import_module(match.group('module'))
[path_planning-27]              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[path_planning-27]   File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
[path_planning-27]     return _bootstrap._gcd_import(name[level:], package, level)
[path_planning-27]            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[path_planning-27]   File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
[path_planning-27]   File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
[path_planning-27]   File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
[path_planning-27]   File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
[path_planning-27]   File "<frozen importlib._bootstrap_external>", line 995, in exec_module
[path_planning-27]   File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
[path_planning-27]   File "/homes/21griepen/colcon_ws/install/bitbots_path_planning/lib/python3.12/site-packages/bitbots_path_planning/path_planning.py", line 11, in <module>
[path_planning-27]     from bitbots_path_planning import NodeWithConfig
[path_planning-27] ImportError: cannot import name 'NodeWithConfig' from 'bitbots_path_planning' (/homes/21griepen/colcon_ws/install/bitbots_path_planning/lib/python3.12/site-packages/bitbots_path_planning/__init__.py)
[ERROR] [path_planning-27]: process has died [pid 936077, exit code 1, cmd '/homes/21griepen/colcon_ws/install/bitbots_path_planning/lib/bitbots_path_planning/path_planning --ros-args -r __node:=bitbots_path_planning --params-file /tmp/launch_params_yrhedkb3 --params-file /tmp/launch_params_0toi1k3l'].
[ball_filter-23] [INFO] [bitbots_ball_filter]: Using frame 'map' for ball filtering (__init__() at /homes/21griepen/colcon_ws/install/bitbots_ball_filter/lib/python3.12/site-packages/bitbots_ball_filter/ball_filter.py:41)
[start_webots_ros_supervisor.py-2] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/supervisor_robot/extern", retrying in 0 second...
[start_single.py-3] [INFO] [webots_ros_interface]: Starting ros interface for amy (__init__() at /homes/21griepen/colcon_ws/install/bitbots_webots_sim/lib/bitbots_webots_sim/start_single.py:23)
[start_single.py-3] Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/amy/extern", retrying in 0 second...
[vision-20] [INFO] [bitbots_vision]: No Profiling available (<module>() at /homes/21griepen/colcon_ws/install/bitbots_vision/lib/python3.12/site-packages/bitbots_vision/vision.py:25)
[localization_handler-26] Traceback (most recent call last):
[localization_handler-26]   File "/homes/21griepen/colcon_ws/install/bitbots_localization_handler/lib/bitbots_localization_handler/localization_handler", line 33, in <module>
[localization_handler-26]     sys.exit(load_entry_point('bitbots-localization-handler==0.0.0', 'console_scripts', 'localization_handler')())
[localization_handler-26]              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[localization_handler-26]   File "/homes/21griepen/colcon_ws/install/bitbots_localization_handler/lib/python3.12/site-packages/bitbots_localization_handler/localization_handler.py", line 62, in main
[localization_handler-26]     dsd = init(node)
[localization_handler-26]           ^^^^^^^^^^
[localization_handler-26]   File "<@beartype(bitbots_localization_handler.localization_handler.init) at 0x73942df56340>", line 32, in init
[localization_handler-26]   File "/homes/21griepen/colcon_ws/install/bitbots_localization_handler/lib/python3.12/site-packages/bitbots_localization_handler/localization_handler.py", line 20, in init
[localization_handler-26]     blackboard = LocalizationBlackboard(node)
[localization_handler-26]                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[localization_handler-26]   File "<@beartype(bitbots_localization_handler.localization_dsd.localization_blackboard.LocalizationBlackboard.__init__) at 0x739429559bc0>", line 32, in __init__
[localization_handler-26]   File "/homes/21griepen/colcon_ws/install/bitbots_localization_handler/lib/python3.12/site-packages/bitbots_localization_handler/localization_dsd/localization_blackboard.py", line 58, in __init__
[localization_handler-26]     self.accel: Float[np.ndarray, 3] = np.array([0.0, 0.0, 0.0])
[localization_handler-26]                 ~~~~~^^^^^^^^^^^^^^^
[localization_handler-26]   File "/homes/21griepen/.local/lib/python3.12/site-packages/jaxtyping/_array_types.py", line 644, in __getitem__
[localization_handler-26]     dim_str = dim_str.strip()
[localization_handler-26]               ^^^^^^^^^^^^^
[... 174 lines cut ...]
[move_group-8] #1    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7efb038effc9, in
[move_group-8] #0    Object "/opt/ros/jazzy/lib/librclcpp.so", at 0x7efb038f4f79, in rclcpp::CallbackGroup::~CallbackGroup()
[move_group-8] Segmentation fault (Address not mapped to object [0x7efaf7ee16e8])
[start_simulator.py-1] running webots --mode=realtime --port=1234 /homes/21griepen/colcon_ws/install/bitbots_webots_sim/share/bitbots_webots_sim/worlds/1_bot.wbt
[ERROR] [move_group-8]: process has died [pid 936058, exit code -11, cmd '/opt/ros/jazzy/lib/moveit_ros_move_group/move_group --ros-args --log-level WARN --ros-args --params-file /tmp/launch_params_3z9k3njd --params-file /tmp/launch_params_xq35dmy1 --params-file /tmp/launch_params_nxrzxydf --params-file /tmp/launch_params_gttwvsuc --params-file /tmp/launch_params_4z2v5q95 --params-file /tmp/launch_params_pb_kes9v'].
[INFO] [aggregator_node-7]: process has finished cleanly [pid 936057]
[INFO] [extrinsic_calibration-11]: process has finished cleanly [pid 936061]
[INFO] [parameter_blackboard-6]: process has finished cleanly [pid 936056]
[INFO] [extrinsic_calibration-10]: process has finished cleanly [pid 936060]
[INFO] [complementary_filter_node-4]: process has finished cleanly [pid 936054]
[INFO] [robot_state_publisher-9]: process has finished cleanly [pid 936059]
[INFO] [motion_odometry-13]: process has finished cleanly [pid 936063]
[INFO] [base_footprint-12]: process has finished cleanly [pid 936062]
[INFO] [odometry_fuser-14]: process has finished cleanly [pid 936064]
[INFO] [animation_node-15]: process has finished cleanly [pid 936065]
[ERROR] [ipm-21]: process has died [pid 936071, exit code -2, cmd '/homes/21griepen/colcon_ws/install/ipm_image_node/lib/ipm_image_node/ipm --ros-args -r __node:=ipm_line_mask --params-file /tmp/launch_params_8l881cyj --params-file /tmp/launch_params_i0awlv4m --params-file /tmp/launch_params_xns0ak29 --params-file /tmp/launch_params_6zf4mknd --params-file /tmp/launch_params_3watlfxx -r camera_info:=/camera/camera_info -r input:=/line_mask_in_image -r projected_point_cloud:=/line_mask_relative_pc'].
[... cut ...] nstall/bitbots_body_behavior/share/bitbots_body_behavior/config/body_behavior.yaml --params-file /homes/21griepen/colcon_ws/install/bitbots_body_behavior/share/bitbots_body_behavior/config/animations.yaml --params-file /tmp/launch_params_n2r90fad --params-file /tmp/launch_params_0ujpkho_ --params-file /tmp/launch_params_18y6jg6z --params-file /tmp/launch_params_6xb1cm6z'].
[INFO] [vision-20]: process has finished cleanly [pid 936070]
[INFO] [HCM-18]: process has finished cleanly [pid 936068]
```

val-ba, 2025-03-27:
Tried to reproduce it. Also happened when you first do colcon build and then colcon build --symlink-install

val-ba, 2025-04-08:
This is probably not an issue on our end, but the problem arises with the generate parameter library, I submitted an issue in their repo: https://github.com/PickNikRobotics/generate_parameter_library/issues/255

closed 2025-04-08 by val-ba

Flova, 2025-04-10:
Thanks for debugging this. We could leave this open until the upstream issue is closed. And if it might persist we would need to work around it in our code.